rizacelik/STM32F405RGT6_INAV_Firmware #3: connecting to mission planner

madamidopingo, 2024-02-07:
hi sir. first thank you for your effort . i have the stm32f405rgt6 dev board (not WE ACT STUDIO ) I have flashed arduplane firmware to it using inav and connected the mpu9250 with spi2 in the dev board when I open mission planner the port appears to me (ardupilot port 9 ) but I cannot connect to it it says no packet received. tried the same connection and  flash inav 7.0 to the same board it worked  .

rizacelik, 2024-02-07:
Which board did you use?

madamidopingo, 2024-02-08:
stm32f405rgt6 devEBox
[image]
[image]

madamidopingo, 2024-02-08:
> Hi sir. first thank you for your effort. I have the stm32f405rgt6 dev board (not WE ACT STUDIO ) I have flashed arduplane firmware to it using inav and connected the mpu9250 with spi2 in the dev [...]

this explanation of what i mean 
[image]
[image]

rizacelik, 2024-02-08:
Is there a stm32f405rgt6 devEBox board pinout diagram? I could not find. If you have a schematic and pinouts, write me and I will help you.

rizacelik, 2024-02-09:
Since the pins of this card and the card we use are used for different purposes, the firmware will not work.
Your card uses PB3, PB4, PB5 pins for FLASH. That's why conflict occurs. You cannot use it.
This will not work in INAV. PB3, PB4, PB5 are motor pins.

[image]

madamidopingo, 2024-02-09:
thanks sir for your help ♥️

rizacelik, 2024-02-09:
I will prepare firmware for your card. It may take a day or two.

madamidopingo, 2024-02-09:
thank you sir very much for helping me , wish you all the best  ♥️

synersignart, 2024-02-15:
can you also have the Target available  for both so I can check as well , thanks Riza oh I may have some info on the H7 hit me up as well

mursa1en, 2024-02-15:
[image]

Hello sir.
I'm using the MPU9250 and connected it over SPI but inav detects  it as the mpu6500. I have tried to select the MPU 9250 manually but it didn't work. Now i have to use the HMC5883. Can you please help me with this issue.

synersignart, 2024-02-16:
BY the way  the SD card doesnt seem to work on the set F405 for some reason

madamidopingo, 2024-02-16:
hi sir , i have a question . PC1 pin in stm32f405rgt6 in the datasheet it's functions are : ETH_MDC/ EVENTOUT / ADC123_IN11 so how it can be used as chip select pin for SPI 2

rizacelik, 2024-02-16:
I haven't been able to complete the firmware yet due to workload. I will definitely prepare it tomorrow. I'm sorry for keeping you waiting.

rizacelik, 2024-02-17:
Today I found time and redefined the pins for you. The compilation process is finished. I couldn't test it because I don't have a board.
If you test it and let me know the result, I can edit it again.

For the Board, you must connect an mpu9250 or mpu6500 and add a BAROMETER. BAROMETER is mandatory.
All Compass sensors are supported.

For Compass and Barometer you need to use I2C1. Pin numbers are given below.
Since there is AK8963 compass inside the mpu9250, you do not need to connect additional compass.

----- Receiver pins ----

PA3 RC In

-------- mpu6500 or mpu9250 sensor pins -------
PB13 SPI2 SCK
PB14 SPI2 MISO
PB15 SPI2 MOSI
PB12 CS

# ------ Compass or Barometer pins ----------

BARO BMP180 (BMP085)
BARO MS56XX
BARO BMP280
BARO BMP388

external All compass types.

PB8 I2C1 SCL
PB9 I2C1 SDA

--- Motor pins ----

PC6  Motor 1
PC7  Motor 2
PC8  Motor 3
PC9  Motor 4

[DeveBoxSTM32F405.zip](https://github.com/rizacelik/STM32F405RGT6_INAV_Firmware/files/14319824/DeveBoxSTM32F405.zip)

rizacelik, 2024-02-17:
[sinersignart](https://github.com/synersignart) 
I will compile for H743. I ordered the board.

madamidopingo, 2024-02-17:
first thank you sir very much , i will flash it now .

madamidopingo, 2024-02-17:
I put the board in dfu mode and flashed the firmware using Inav configurator 7 until a successful message appeared then I put the board in flash mode and connected mpu9250 to the mapping that you provided sir, but the same problem appeared.

rizacelik, 2024-02-17:
Which sensors did you use?

rizacelik, 2024-02-17:
MPU650 or MPU9250 Sensor pins connection.

[image]

madamidopingo, 2024-02-17:
thank you sir very much. I am very sorry. I checked the connection but no gain, I don't know what is the problem.

rizacelik, 2024-02-17:
Since I don't have the board, it is difficult to edit and test the software. Let's wait for others to try it too. I hope we can find a solution.

mursa1en, 2024-02-18:
[image]

I'm using the same board and both firmware works fine. The problem I'm facing on mission planner is that i need to reinstall it every time or the doesn't connect.

rizacelik, 2024-02-18:
I don't have this board. I compiled it for you, but I was told it didn't work. Since there is no board, I have to do the compilation by default. There's nothing I can do about this.

madamidopingo, 2024-02-21:
> ![IMG_20240216_175855.jpg](https://private-user-images.githubusercontent.com/108753302/305715709-29567ba3-731c-4fbf-b2fa-42d386b52fe1.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDg1MzcwNTIsIm5iZiI6MTcwODUzNjc1MiwicGF0aCI6Ii8xMDg3NTMzMDIvMzA1NzE1NzA5LTI5NTY3YmEzLTczMWMtNGZiZi1iMmZhLTQyZDM4NmI1MmZlMS5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMjIxJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDIyMVQxNzMyMzJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT03MDU4MmJkOWY3NzBjNzNmM2YyNTQwY2RiMDFiYTgxNTE1ZTljNWU5ZDg4NjQ1YjBlYmY5OTczN2Q4Mjc1OTIzJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.OyUQ40iyHzBsds7N4rEnYQeg0W-wpkfnNJJajc7oOsM)
> 
> I'm using the same board and both firmware works fine. The problem I'm facing on mission planner is that i need to reinstall it every time or the doesn't connect.

hi mursa1en i think you are flashing firmware using usb to ttl because when you are on bootloader mode you can flash the firmware but you need to convert to flash mode before you remove the device cause the firmware will be erased

madamidopingo, 2024-02-21:
I am flashing the firmware in dfu by using inav it flashed successfully but it doesn't connect to mission planner

rizacelik, 2024-02-22:
I recompiled it. Can you install it on the board and test it?

[DeveBoxSTM32F405.zip](https://github.com/rizacelik/STM32F405RGT6_INAV_Firmware/files/14371307/DeveBoxSTM32F405.zip)

madamidopingo, 2024-02-22:
> I recompiled it. Can you install it on the board and test it?
> 
> [DeveBoxSTM32F405.zip](https://github.com/rizacelik/STM32F405RGT6_INAV_Firmware/files/14371307/DeveBoxSTM32F405.zip)

thank you sir very much ,I installed it but the same problem appear .I appreciate your help , wish you all the best , can you please share hwdef for this board , so I can try with it .

rizacelik, 2024-02-22:
hwdef for this board
[hwdef.zip](https://github.com/rizacelik/STM32F405RGT6_INAV_Firmware/files/14378553/hwdef.zip)

mursa1en, 2024-03-17:
> Since the pins of this card and the card we use are used for different purposes, the firmware will not work.
> Your card uses PB3, PB4, PB5 pins for FLASH. That's why conflict occurs. You cannot use it.
> This will not work in INAV. PB3, PB4, PB5 are motor pins.
> 
> [image]
> 

Will it work if I remove this component from the board? Sorry for this dumb question sir. I want to make this but the board you've used is not available here

rizacelik, 2024-03-17:
https://www.aliexpress.com/item/1005006051651136.html
STM32F405RGT6 board is very cheap. It is 50% cheaper than the card you have. I think you should get this. I can't compile software because I don't have your board.

NOTE: Since FLASH is embedded, you cannot remove it. Board still doesn't work. Don't try. The board will be damaged.

madamidopingo, 2024-03-17:
> https://www.aliexpress.com/item/1005006051651136.html STM32F405RGT6 board is very cheap. It is 50% cheaper than the card you have. I think you should get this. I can't compile software because I [...]

thanks sir very much . i have already buy it .

mursa1en, 2024-03-18:
> https://www.aliexpress.com/item/1005006051651136.html
> STM32F405RGT6 board is very cheap. It is 50% cheaper than the card you have. I think you should get this. I can't compile software because I [...]

mursa1en, 2024-03-18:
> https://www.aliexpress.com/item/1005006051651136.html
> STM32F405RGT6 board is very cheap. It is 50% cheaper than the card you have. I think you should get this. I can't compile software because I [...]

Thanks a lot for your response. 
I don't have access to AliExpress and that board is not available in the local market.

Also I should have listened to you. Now the board only draws tons of current and heats up. 

I will be making the f411 now.

skimans, 2024-04-04:
> thank you sir very much ,I installed it but the same problem appear .I appreciate your help , wish you all the best , can you please share hwdef for this board , so I can try with it .

Hi @rizacelik thanks much for your excelent work with this board.
I want ask wich external compass i can use with this firmware.
MPU i want use MPU6500 and GPS i have Beitian 880 that use HMC5883 or QMC5883 compass.
These two compass are already in your firmware ?

thanks much

skimans, 2024-04-04:
> Thanks a lot for your response. I don't have access to AliExpress and that board is not available in the local market.
>
> Also I should have listened to you. Now the board only draws tons of current [...]

hi @mursa1en for STM32F411nceu  work ??

skimans, 2024-04-04:
> Thanks a lot for your response. I don't have access to AliExpress and that board is not available in the local market.
>
> Also I should have listened to you. Now the board only draws tons of current [...]

hi @mursa1en F411 board work ??
i think have problem of low memory..

skimans, 2024-04-09:
@madamidopingo do you have solved ? your board connect now to Mission Planner ?

mursa1en, 2024-04-10:
> hi @mursa1en F411 board work ??
> i think have problem of low memory..

I'm using Mr Celik's firmware for the f411 and my quad is flying without any issues although it needs some tuning.

skimans, 2024-04-10:
> I'm using Mr Celik's firmware for the f411 and my quad is flying without any issues although it needs some tuning.

@mursa1en Excelent news!!
these vening i try to install firmware on my STM32F411NCEU board.

then do you have used this firmware...right ??

https://github.com/rizacelik/STM32F411CEU6_INAV_Firmware

thanks for info

mursa1en, 2024-04-10:
> @mursa1en Excelent news!!
> these vening i try to install firmware on my STM32F411NCEU board.
>
> then do you have used this firmware...right ??
>
> https://github.com/rizacelik/STM32F411CEU6_INAV_Firmware [...]

Yes. I used the inav 7.0https://github.com/rizacelik/STM32F405RGT6_INAV_Firmware/assets/108753302/ef250e0a-54c5-44a1-8050-d1a38ced7c34

skimans, 2024-04-10:
> Yes. I used the inav 7.0https://github.com/rizacelik/STM32F405RGT6_INAV_Firmware/assets/108753302/ef250e0a-54c5-44a1-8050-d1a38ced7c34

@mursa1en then you dont installed Ardupilot on F411 ?
but only Inav...

mursa1en, 2024-04-10:
> > > > > > > https://www.aliexpress.com/item/1005006051651136.html
> > > > > > > STM32F405RGT6 board is very cheap. It is 50% cheaper than the card you have. I think you should get this. I can't compile software because I don't have your board.
> > > > > > > NOTE: Since FLASH is embedded, you cannot remove it. Board still doesn't work. Don't try. The board will be damaged.
> > > > > > 
> > > > > > 
> > > > > > Thanks a lot for your response. I don't have access to AliExpress and that board is not available in the local market.
> > > > > > Also I should have listened to you. Now the board only draws tons of current and heats up.
> > > > > > I will be making the f411 now.
> > > > > 
> > > > > 
> > > > > hi @mursa1en F411 board work ??
> > > > > i think have problem of low memory..
> > > > 
> > > > 
> > > > I'm using Mr Celik's firmware for the f411 and my quad is flying without any issues although it needs some tuning.
> > > 
> > > 
> > > @mursa1en Excelent news!!
> > > these vening i try to install firmware on my STM32F411NCEU board.
> > > then do you have used this firmware...right ??
> > > https://github.com/rizacelik/STM32F411CEU6_INAV_Firmware
> > > thanks for info
> > 
> > Yes. I used the inav 7.0https://github.com/rizacelik/STM32F405RGT6_INAV_Firmware/assets/108753302/ef250e0a-54c5-44a1-8050-d1a38ced7c34
> 
> @mursa1en then you dont have installed Ardupilot on F411 ?
> but only Inav...
> 

The F411 theoretically can't run ardupilot. 🫥
I guess inav offers more than my requirements

mursa1en, 2024-06-07:
@rizacelik 

, I apologize for bothering you again. I’ve found this board at the local market. Could you please review its schematic and tell me if the firmware is meant to work with this board? It would also be very helpful if you could share the schematic for the WeAct bo

[image]

[image]

[image]

[image]

rizacelik, 2024-06-07:
Install STM32F405RGT6 Firmware, it should work.

rizacelik, 2024-06-07:
An Acceleration and Gyroscope module must be connected to the board. It will not connect to the Bare Board Configurator.

rizacelik, 2024-06-07:
the schematic for the WeAct board
https://github.com/WeActStudio/WeActStudio.STM32F4_64Pin_CoreBoard/blob/master/Hardware/WeAct-STM32F4_64PIN-CoreBoard_V10%20SchDoc.pdf

skimans, 2024-09-12:
@madamidopingo i have same board of you. I cannot connect board to Mission Planner. Do you have solved ? you can connect it ? wich version work fine ? thanks

madamidopingo, 2024-09-12:
> @madamidopingo do you have solved ? your board connect now to Mission Planner ?

> @madamidopingo i have same board of you. I cannot connect board to Mission Planner. Do you have solved ? you can connect it ? which version work fine ? thanks

no , i buyed weact stm32f405 and worked perfectly  
thanks @rizacelik

closed 2024-12-01 by rizacelik