rizacelik/STM32F405RGT6_INAV_Firmware #4: Not making com port with arducopter firmware.

LalitKandpal12, 2024-02-28:
Good evening sir, I were trying to put your arducopter firmware in my stm32board (stm32f407vet6) but after installing it to my stmboard it was not able to make any com port. But when i installed your inav firmware in my board it was running perfectly. My board is slightly different then you targeted board, but it was able to run your inav firmware , also my board has 1 mb flash. Can there be any way that it can run on my board also?

rizacelik, 2024-02-28:
Have you read the problem here https://github.com/rizacelik/STM32F405RGT6_INAV_Firmware/issues/3 ? Is your card the same? If not, can you provide the link or information of your card, if you have one?

LalitKandpal12, 2024-02-29:
Thank you for your response. I have a slightly different board, the STM32F407VET6. Although the pins on both boards seem identical, as it was able to run iNav with the same pin configuration you provided, I encountered an issue after installing the arducopter firmware. Upon connecting it to my laptop, it failed to create any COM port.

this is my board - https://stm32-base.org/boards/STM32F407VET6-STM32-F4VE-V2.0.html (my board has 1 mb flash).

rizacelik, 2024-02-29:
`W25Q16JV pins
#	Name	Function	Connected to
1	-	/CS	PB0
2	-	DO	PB4
3	-	/WP	+3.3V rail
4	-	GND	Ground plane
5	-	DI	PB5
6	-	CLK	PB3
7	-	/HOLD	+3.3V rail
8	-	VCC	+3.3V rail`

PB3, PB4, PB5, PB0 pins will not work because they are used by W25Q16JV flash memory.
Since I do not have this board, it is not possible to create firmware for it. It is necessary to prepare by constantly testing. I wish I could help you, but unfortunately I can't.

LalitKandpal12, 2024-02-29:
Thanks for your quick guidance. If you have a moment to spare, I'd really appreciate your guidance on how can i compile my own version for board. Totally understand if you don't have time though!.
Thanks again for helping : )