rizacelik/STM32F405RGT6_INAV_Firmware #8: Please provide target build .c/.h files

xznhj8129, 2024-07-01:
Hello, i need the src/target header files to compile the new INAV versions to this board. I think you are also not supposed to provide compiled firmware without also providing the code used, due to the license. It would be very appreciated if you could provide them.

mursa1en, 2024-07-01:
> Hello, i need the src/target header files to compile the new INAV versions to this board. I think you are also not supposed to provide compiled firmware without also providing the code used, due to the license. It would be very appreciated if you could provide them.

You can use the files for the Matek f405 wing as a reference. It has support for the mpu6500, bmp280 amd other complementary components. Also you can edit it. I don't know if it is helpful to you or not. 

You may find it helpful.
https://www.youtube.com/live/5wlr4WYpv1w?si=Gq8EPGZK3kLzQ4Iw

closed 2024-12-01 by rizacelik

shrinibas7, 2025-02-03:
Hey riza the board you are using looks like f407 can i use f407 ?

rizacelik, 2025-02-03:
It has similar hardware, you can use it. Are there any details about the card?  should look.

shrinibas7, 2025-02-03:
> It has similar hardware, you can use it. Are there any details about the card? should look.

I already made a f405 board but after buying chip it’s not going dfu and there are some pcb issues so i made another one ,i was wondering if there are any firmware to upload and then i can go dfu..will be a great help thank you..