rizacelik/STM32F405RGT6_INAV_Firmware #9: Motor 4 is not working (PWM) + support og GY-91

jct1411, 2024-07-14:
I setup some config with WeAct Studio STM32F405RGT6 Board but motor 4 (PWM) does not work.  I tried with 2 boards and also probed the signal with an oscilloscope but not signal is shown.  

It would also be great if GY-91 could be supported since barometer is not recognized when cabling with SPI.  

Thanks

mursa1en, 2024-08-01:
I am also facing the same problem on the motor 4. Were you able to resolve it?

mursa1en, 2024-08-02:
> I setup some config with WeAct Studio STM32F405RGT6 Board but motor 4 (PWM) does not work.  I tried with 2 boards and also probed the signal with an oscilloscope but not signal is shown.
>
> It would [...]

Which firmware did you use? Motor 4 is working fine on the arducopter firmware.

mursa1en, 2024-08-02:
Motor 4 is also working on inav 6.1.1

jct1411, 2024-08-07:
I used Inav 7.0.0.  I did not try with firmware 6.1.1

jct1411, 2024-08-07:
I was not able to resolve the problem but it seems to be related to some invalid timer which is assigned to the pin corresponding to motor 4.  Not possible apparently to solve such issue via CLI in Inav and some rebuild is probably mandatory.

closed 2024-12-01 by rizacelik