rizacelik/STM32F405RGT6_INAV_Firmware #10: “Multiple Issues with MPU6500, SD Card, and Motor Output on iNav 7.0 and ArduCopter Firmware”

mursa1en, 2024-08-01:
Hello Sir,

I hope you are doing well. I have a board similar to the WeAct one. I have installed the MPU6500 over SPI, and the QMC5883 and BMP280 over I²C. I am facing some issues:

On iNav 7.0, the SD card doesn’t seem to work, but it works on ArduCopter firmware.

Motor 4 does not seem to give any output on Multishot and DShot protocols (I could not test it on PWM as my ESC doesn’t support it).

Additionally, I am facing a problem where the MPU6500 is being recognized as an MPU6000 on ArduCopter firmware, while it is recognized correctly as the MPU6500 on iNav firmware.

In the schematics, the connections for all hardware and software serial ports are not mentioned. It would be a great help if you could share the full schematics.

I also have some requests:

Please add support for octocopters (8 motors).
Support for VL53L1X/VL53L0X rangefinders.
Thank you.
[image]

mursa1en, 2024-08-02:
[image]

skimans, 2024-09-11:
@mursa1en hi. wich version of firmware do you have installed in your board. I'm have same of your board.
do you have link for download it ?
thanks much

skimans, 2024-09-12:
@mursa1en ok. but wich version do you have installed with Mission Planner ? becouse i cannot connect my board in MP. i use MPU6500 but not connect.

skimans, 2024-10-08:
@mursa1en in wich pin do you have connected MPU6500 ? i have connected in SPI2 pins..the sames indicated from Riza...but MPU6500 is not recognized...

skimans, 2024-10-09:
@rizacelik for my problem above..in Inav these connections works fine and MPU6500 is recognized but in Ardupilot i connect in Mission Planner but MPU6500 is not recognized.