rizacelik/STM32F405RGT6_INAV_Firmware #12: BetaFlight

huseynmammad, 2024-11-15:
Merhaba Hocam. Öncelikle proje için teşekkür ederim. Bizim devremiz Betaflight için INAV 7.1'deki devreyle aynı mı? Ayrıca, MPU6050 I2C ile olan devrede Betaflight MPU6050'yi görmüyor. Sanırım MPU9250 ile olan şemayı kullanmam gerek. Bu konuda yardımcı olursanız çok sevinirim. Şimdiden teşekkürler, Hocam.

rizacelik, 2024-11-16:
Betaflight için maalesef derleme dosyası yok.

huseynmammad, 2024-11-16:
Hocam son youtube videonuzda BetaFlight kullanıyorsunuz.  Peki sizde hangi devreyle çalışıyor bu? Teşekkürler

rizacelik, 2024-11-17:
Buraya bakın. https://github.com/rizacelik/STM32F411CEU6_INAV_Firmware

ayerko40, 2024-11-26:
Hocam çok rica ediyorum mümkünse bunun Betaflight için olanını derleme yapabilir misiniz?

Saygı ve Sevgilerimle

closed 2024-12-01 by rizacelik