rizacelik/STM32F405RGT6_INAV_Firmware #14: MSP connection closes immediately with Inav 7.0 on WeAct Studio STM32F405RGT6 with GY-87 only

jct1411, 2024-12-02:
Dear Sir,

I setup some config with WeAct Studio STM32F405RGT6 Board on a breadboard with GY-87 only.  I flashed the board with the provided Inav 7.0 firmware.  When trying to connect with Inav configurator 7.0.1 or 7.1.0 ot 7.1.1, MSP connection closes immediately.  This cannot be related to the component GY-87 since the same physical component shows up correctly using your Inav firmware for STM32F411 (also breadboard with GY-87 only) .  Any idea what the problem could be ?

[image]

Thanks in adance for your help

rizacelik, 2024-12-02:
Use a sensor other than GY-87. GY-87 also works with MPU6050 I2C and is quite slow. This sensor is no longer supported.

jct1411, 2024-12-02:
I presume a new build of the Inav firmware is required which supports GY-91 ?  How do I connect GY-91 ?  Same cabling as for Betaflight ?  But you mention that this the circuit connections is for betaflight ...

Thanks

jct1411, 2024-12-02:
Have tried with GY-91 connected as with Betaflight (now that GY-91 is correctly connected : see my issue concerning Betaflight) but the issue with Inav remains the same (MSP connection closes immediately).  May be some issue with this old Inav firmaware ?

rizacelik, 2024-12-02:
There is no GY-91 support for INAV at the moment. If I find time, I will add support. Unfortunately, I don't have time.

jct1411, 2024-12-02:
I understand.  Note however that I have been using Inav 7 with GY-91 on STM32F411 ...