rizacelik/STM32F405RGT6_INAV_Firmware #15: Arducopter on WeAct Studio STM32F405RGT6 with GY-87 only:  "config error : unable to initialize driver" and QuadX not recognized

jct1411, 2024-12-02:
Dear Sir,

I setup some config with WeAct Studio STM32F405RGT6 Board on a breadboard with GY-87 only.  I flashed the board with the provided Arducopter firmware.  Connection  with latest stable version of Mission Planner is working but  I get " config error : unable to initialize driver" and going into the setup QuadX is not recognized.  What could be the problem ?

[image]
[image]

Thanks for your help

rizacelik, 2024-12-02:
Use a sensor other than GY-87. GY-87 also works with MPU6050 I2C and is quite slow. This sensor is no longer supported.

jct1411, 2024-12-02:
I presume a new build of the Arducopter firmware is required which supports GY-91 ?  How do I connect GY-91 ?  Same cabling as for Betaflight ?  But you mention that this the circuit connections is for betaflight

Thanks

jct1411, 2024-12-02:
I connected GY-91 the same as for Betaflight.  When connected, Mission Planner 1.3.82 reports now explicitely "Config Error : Baro: unable to initialize driver" and "Config Error: fix problem then reboot".  
QuadX is still not recognized.

rizacelik, 2024-12-02:
There is no GY-91 support for Arducopter at the moment. If I find time, I will add support. Unfortunately, I don't have time.

shrinibas7, 2025-03-15:
> Use a sensor other than GY-87. GY-87 also works with MPU6050 I2C and is quite slow. This sensor is no longer supported.

Hey riza ,i added bmi270 works fine in inav but saying config error ins in ardupilot pls help…

synersignart, 2025-04-24:
[image]
imusing the BMI160 HMC5883 and BMP280
also getting configure error

shrinibas7, 2025-04-24:
> [image] imusing the BMI160 HMC5883 and BMP280 also getting configure error

Hey synerflight🇵🇭 same thing happened with me bmi270 worked with inav but not with ardupilot…

synersignart, 2025-05-05:
> > [image] imusing the BMI160 HMC5883 and BMP280 also getting configure error
> 
> Hey synerderg🇵🇭 same thing happened with me bmi270 worked with inav but not with ardupilot…

yup  strange because I also have my IMU on ISP2 and the sensors Riza listed are also supported