rizacelik/STM32F405RGT6_INAV_Firmware #16: Latest Betaflight firmware : Baro and Mag not found for GY-91 and mixer mode issue in "motors"

jct1411, 2024-12-02:
Dear Sir,

I have setup some config with WeAct Studio STM32F405RGT6 Board on a breadboard with GY-91 only.  I flashed the board with the latest Betaflight firmware of 01 dec 2024.  Gyro and Accel are well recognized but not Baro nor Mag ?  Baro and Mag of the same physical GY-91 are well recognized in a similar configuration with Betaflight firmware 4.5.0 on STM32F411

Also going into "motors" there is popup with "Mixer mode problem detected".  Looking at resources with CLI, I only see Motor 1 and Motor 2.  I do not see anything for Motor 3 and Motor 4 (B06 and B07 are also marked as free)

[image]
[image]
[image]

Thanks for your help

jct1411, 2024-12-02:
Strange, same physical GY-91 module with STM32F411 shows Baro (not Mag).  I only have one GY-87 module where the compass is working ...   I will check again the connections for GY-91

Strange that I do not find the resources for Motor 3 and Motor 4.  Can I add these manually ?  Why don't they appear automatically after loading the firmware ?

rizacelik, 2024-12-02:
You are right, there is a conflict between MOTOR 1 TIMER and DMA and it has been resolved. The connection diagram has been recreated. You can test it.

jct1411, 2024-12-02:
Thank you for the modification : popup does no more appear when selecting "motors".  However I don't measure any signal on any of the motors' pins.

rizacelik, 2024-12-02:
You can test it by giving output on this screen. You should get a signal.

[image]

jct1411, 2024-12-02:
This is the method I have used but with PMW.  I have again tested this method on another configuration with STM32F411 and I immediately see the signals but on the breadboard configuration with STM32F405RGT6 no signal is seen (I use an oscilloscope).  There must be an issue somewhere ...

rizacelik, 2024-12-03:
Problem fixed. All motors are working. Download and testing.

jct1411, 2024-12-03:
Motor 1, Motor 2 Motor 3 are working .  **However Motor 4 does not**.  I see Motor 4 is the only one using TIM3 CH2.  May be there is the problem.  I rememeber a few months ago having looked at such an issue but I can no more remember what exactly the issue was.  I found this by lookink at the technical documentation of the processor but I do no more find it ...

I also tried UART 1 with GPS.  It doesn't seem to work.    Betaflight does not detect the GPS when it is connected (icon GPS does not appear, even red when just connected).  Il will check the GPS on another configuration with STM32F411

jct1411, 2024-12-03:
I confirm indeed there seems to be an issue with GPS UART 1 since ithe same physical GPS works on ST32M32F411 config but not on STM32F405RGT6 config ...

rizacelik, 2024-12-03:
I disabled the servos, 4 motors work but the GPS did not work. I tried very hard but it did not work. I am removing the firmware. I will recompile it at a later time.

jct1411, 2024-12-03:
Thanks a lot for all the work already done.  I know how this can be time consuming.
Waiting for a new firmware version that I will test as soon as available.

Flighthobbystm, 2025-06-19:
> Problem fixed. All motors are working. Download and testing.

Where can I download Betaflight ?? fixed??